evothings/cordova-ble #51: readCharacteristic() doesn't return fail callback

alexisduque, 2015-01-21:
When I call ``readCharacteristic() `` on a peripheral implementing PASSKEY security, Android often doesn't ask for passkey.

So ``readCharacterstic`` failed but neither success nor error callback are called.
However, using logcat, I can see that the API returns GATT_INSUFFICIENT_AUTHENTICATION (status = 5)
``D/BluetoothGatt(23821): onCharacteristicRead() - Device=FD:8B:BA:4B:1D:27 UUID=f3851521-5942-b2cb-ae40-271cd0dd17e5 Status=0``

fredrikeldh, 2015-02-03:
I think this a bug in Android; my [handler](https://github.com/evothings/cordova-ble/blob/21f9ad3904887d239001ad89b4b7bb22c53775b0/src/android/BLE.java#L579) calls the error callback when it gets any status except SUCCESS.

fredrikeldh, 2015-02-03:
On second thought, we might add a timeout of our own. Wouldn't report the proper error code, but at least it would report *some* error.

mikaelkindborg, 2016-11-24:
There is now support for explicit paring/bonding in the BLE plugin, that should help with this issue.

There are however reports of that this works a bit unstable, not sure if this is the plugin or the Android implementation. See this issue: https://github.com/evothings/cordova-ble/issues/129

Closing this issue, let me know if to reopen.

closed 2016-11-24 by mikaelkindborg