evothings/cordova-ble #53: Code running in android but not on ios

shraddha1112, 2015-02-26:
Hello,
I am doing a project where we are connecting to our custom device and trying to communicate over BLE. We have done succesful round of implementation and test using your plugin with Android. Now I have to test same with ios. I am new to ios. I build and installed same application for ios. I can scan and connect using this plugin. But rest is not working. Not even giving error. Dont understand what is going wrong:

  if(zDeviceID != null){
        console.log('Got some device id '+zDeviceID);
        evothings.ble.connect(zDeviceID,connectSuccess,failure);
}
 var connectSuccess = function(e){
          console.log("Connect Success");
          zenDeviceHandle = e.deviceHandle;
evothings.ble.services(zenDeviceHandle,onServiceReadSuccess,failure);
}
var onServiceReadSuccess = function(services)
{
console.log("Read service done");
console.log("Services "+JSON.stringify(services));
var serviceHandle;
for (var i = 0; i < services.length; i++)
    {
      var service = services[i];
      console.log('BLE service: ');
      serviceHandle = service.handle;

      console.log('  ' + service.handle);
      console.log('  ' + service.uuid);
      console.log('  ' + service.serviceType);
    }
   evothings.ble.characteristics(1,serviceHandle,onCharReadSuccess,failure);

}
After this evothings.ble.services call nothing happens. Where as when I run same on android i get list of services and interanlly get the list of charachteristics.

MAC version : version 10.8.5
xcodebuild -version : 5.1
cordova -version : 4.2.0
Ios version 7.0.1

Can some one please help me? 

Regards,
Shraddha

shraddha1112, 2015-02-26:
On debugging into ios code we got to know that the services function in EVOBLE.m is getting command as null. i.e. parameter (command) is coming null..

fredrikeldh, 2015-02-26:
Cordova versions >= 4.0.0 are not yet supported by this plugin. I'll update plugin.xml to indicate that.

You might try [Evothings Studio](http://evothings.com/) for development. It has a Client app, precompiled and free on the App Stores, with the BLE plugin included.

shraddha1112, 2015-02-27:
Thanks a lot for the response. I will try with older version.

mikaelkindborg, 2016-07-10:
Does this problem persist? Closing issue for now. let me know if to reopen.

closed 2016-07-10 by mikaelkindborg