evothings/cordova-ble #54: ios: Incompatible pointer types sending 'NSObject *' to parameter of type 'NSDictionary *'

mzarella, 2015-03-05:
Hi, I opened the cordova project in xcode: Command+O, then select my_project/platforms/ios as the project root

I can emulate the app by selecting Product -> Run from the file menu. After verifying the app is running in the emulator, I attempted to build the project (Product -> Build), but I received an error:

/Users/mario/MyProject/platforms/ios/MyProject/Plugins/com.evothings.ble/EVOBLE.m:1295:22: Incompatible pointer types sending 'NSObject *' to parameter of type 'NSDictionary *'

Suggestions on how to get around this?

divineprog, 2015-03-06:
Hi mzarella, thanks for reporting. This problem is causes by an implicit downcast. Submitted a pull request with a fix: https://github.com/evothings/cordova-ble/pull/55

fredrikeldh, 2015-03-06:
Fixed by #55.

closed 2015-03-06 by fredrikeldh