evothings/cordova-ble #56: one device is being polled more than the others

userfive, 2015-03-08:
I'm trying to find the closest beacons, and I've ran into an issue where one device is being found more than the others. In the following:

```
evothings.ble.startScan(function(device) {
  if (device.rssi <= 0) callbackFun(device, null)
```

`device.address` tends to not change. Has this issue been discovered before, and how did you get around it? If it hasn't been seen before, what are some ways that I might get around it?

Thank you.

userfive, 2015-03-09:
To make it a little more clear, tne device is being found way, way more than the others, so it isn't giving me an accurate reading of which devices are closest since the rssi of other devices is rarely read.

fredrikeldh, 2015-03-09:
BLE scanning does not work by polling. The scanner is a passive listener that listens to advertisements from the scannees. Your "one device" may be set to advertise more often than the others. Alternatively, it may be closer to your scanning device, so that advertisements from the other devices are being lost in radio noise.

It is also worth mentioning that BLE and WiFi operates on the same 2.4Ghz band, and will interfere with each other.

closed 2015-03-10 by userfive