evothings/cordova-ble #62: Consider adding scanRecord parsing to the native code

thegecko, 2015-04-17:
The scanRecord parsing which appears in the easyble.js code would be so much more useful if it was added to the native code layer by default.

advertisementData and scanRecord data could then be accessed in a common way, perhaps even replacing the existing structures.

This would remove the need to (slowly) parse the data in JavaScript and make the API more consistent across devices.

BioPhoton, 2015-11-23:
1+

mikaelkindborg, 2016-11-24:
Parsing of scan record on Android is now added the the BLE plugin, in JavaScript, not natively. Performance seems to be satisfactory. Let me know if there are needs for enhancement of performance or other issues, we can then revisit this issue and reopen.

closed 2016-11-24 by mikaelkindborg