evothings/cordova-ble #64: Reading a different UUID depending on which phone is used

mzarella, 2015-04-20:
https://forums.estimote.com/t/reading-more-than-one-uuid/1067

The callback to evothings.ble.startScan, `function deviceFound(device)`, is reading different addresses, i.e. `device.address` is different depending on the phone. Has anyone ever ran into this?

fredrikeldh, 2015-04-20:
Yeah, that's a result of how the iOS CoreBluetooth API works. It hides the MAC address, and instead assigns a UUID to each BLE device it detects. These UUIDs are random and temporary; you run the same program again on the same device, you might get different ones.

closed 2015-04-20 by fredrikeldh

mzarella, 2015-04-20:
@fredrikeldh But then how can I associate a randomly generated address with an object in the DB? Or rather, what's the proper way to associate a device with the information you want to present? It doesn't seem like device.address is going to be the way to do this.

fredrikeldh, 2015-04-21:
"proper" way? That's up for debate. If you have control over the devices' advertisements, you might add some Manufacturer Data, or some Service Data, which you can then read through ```device.advertisementData```.

I see you've mentioned Estimote, so I'll guess you're using iBeacons. In that case, I recommend using one of our [iBeacon](https://github.com/evothings/cordova-plugin-ibeacon) [plugins](https://github.com/evothings/phonegap-estimotebeacons). They will give you more permanent device identifiers.