evothings/cordova-ble #67: IOS background connect not connecting after a while

arickatz, 2015-05-18:
I am working with the plugin on android and ios.
On ios the scan is not working in the background although all plist values are set so
I am using the connect function that works correctly in the background.
I have a solution that tries to connect to a near by estimate beacon every 30 seconds during a time span of 30 minutes to 2 hours, mostly runs in the background. for some reason on some iphone devices after a while, can happen during a phone call when app is in the background, maybe due to cpu usage, the connect function fails to connect. I tried to connect and disconnect 10 times every 10 or 30 seconds but I get no response and also if the app returns to the foreground it still cant scan (that works in the foreground), only if I restart the app connection can be established. any idea why? and how can I overcome this issue? Would this happen if I would convert to the estimote plugin? 
are there different functions that I can use like advertise or notify that may be more reliable? apparently the app works correctly in the background because it tries to connect multiple times and stops the app and sends a notification that more than 10 times the ble device was not connectable.

mikaelkindborg, 2016-03-21:
@arickatz Sorry for the very slow response. What is the status of your work?

In the process of adding background BLE support for iOS. 

See this branch: https://github.com/mikaelkindborg/cordova-ble/tree/background

Not sure how your use case wityh connect/disconnect in the background would behave. Scanning however works in the background in iOS with the code in the background branch. This will be merged into the master branch.

mikaelkindborg, 2016-03-22:
Support for iOS background scanning now merged into the master branch.

mikaelkindborg, 2016-03-22:
@arickatz Closing this issue, contact me if you wan to reopen, email mikael@evothings.com

closed 2016-03-22 by mikaelkindborg