evothings/cordova-ble #72: writeWithoutResponse

fredrikeldh, 2015-07-21:
We would like the ability to write characteristic data without waiting for the remote device to respond, thus increasing data throughput.

I think there are two ways to do this: a function setWriteType(), matching the [Android API](http://developer.android.com/reference/android/bluetooth/BluetoothGattCharacteristic.html#setWriteType%28int%29), or a function writeCharacteristicWithoutResponse(), which calls setWriteType behind the scenes, so to speak.

Relevant links:
http://evothings.com/forum/viewtopic.php?f=8&t=1479
http://stackoverflow.com/questions/24135682/android-sending-data-20-bytes-by-ble
https://github.com/fishkingsin/BLEDialogTool/blob/master/src/com/fishkingsin/ble/dialogtool/BLEService.java

fredrikeldh, 2015-11-23:
How could this be implemented on iOS?

fredrikeldh, 2015-11-23:
Android implementation here: https://github.com/fredrikeldh/cordova-ble/tree/writeWithoutNotification

mikaelkindborg, 2016-03-22:
Android implementation merged into master branch.

TODO: Implement on iOS (is supported) and test.

mikaelkindborg, 2016-07-05:
Updated EasyBLE library with function writeServiceCharacteristicWithoutResponse.

Documentation comment:

	/**
	 * Write value of a characteristic for a specific service without response.
	 * This faster but not as fail safe as writing with response.
	 * Asks the remote device to NOT send a confirmation message.
	 * Experimental, implemented on Android.
	 * @param {string} serviceUUID - UUID of service that has the characteristic.
	 * @param {string} characteristicUUID - UUID of characteristic to write to.
	 * @param {ArrayBufferView} value - Value to write.
	 * @param {evothings.easyble.emptyCallback} success - Success callback: success().
	 * @param {evothings.easyble.failCallback} fail - Error callback: fail(error).
	 * @public
	 * @instance
	 * @example
	 *   device.writeServiceCharacteristicWithoutResponse(
	 *     serviceUUID,
	 *     characteristicUUID,
	 *     new Uint8Array([1]), // Write byte with value 1.
	 *     function()
	 *     {
	 *       console.log('BLE data sent.');
	 *     },
	 *     function(errorCode)
	 *     {
	 *       console.log('BLE writeServiceCharacteristicWithoutResponse error: ' + errorCode);
	 *     });
	 */

mikaelkindborg, 2016-07-26:
Implemented writeCharacteristicWithoutResponse on iOS in this commit: https://github.com/evothings/cordova-ble/commit/4fd629080e1e4aa3a5a7ecc95b3ea35597a1a752

Made new release of plugin: https://github.com/evothings/cordova-ble/releases/tag/1.4.3

Function writeCharacteristicWithoutResponse tested working on iOS and Android.

Closing issue.

closed 2016-07-26 by mikaelkindborg