evothings/cordova-ble #74: Problems with BLE under Android 5.1.1

RolfKH, 2015-08-06:
Hello,
We have an application that was fine running under Android 4.4.2.
But with Android 5.1.1 we are getting connection problems, timeout on connection and unstable Bluetooth (switches on and of).
We have updated the Cordova library.

Do you know of any issues with the newer Android operating system?

Regards,
Rolf

fredrikeldh, 2015-08-07:
I have not encountered this myself, though I have a vague memory of seeing another similar report. What hardware are you using? I have a Nexus 6.

RolfKH, 2015-08-07:
> Dato: 07.08.2015 17:39 (GMT+01:00)
> Ko: Rolf Kahrs Hansen <rolf@spiromedical.com>

samsung s6

Sendt fra min Samsung Galaxy-smarttelefon.

-------- Opprinnelig melding --------

fredrikeldh, 2015-08-10:
I don't have an S6, I don't have your app, I don't have whatever BLE device the app is talking to, and the problem is intermittent. Tricky. :/

Perhaps you can try some other phone. Also it would be helpful if you would try some of the [Evothings](http://evothings.com/) [examples](http://evothings.com/doc/examples/examples.html).

RolfKH, 2015-08-10:
We will research this further and get back

RolfKH, 2015-09-22:
Hi
We changed the delay times (-> shorter) on the device and that helped!

Another question: Can Cordova (with BLE) be ported to Windows?

fredrikeldh, 2015-09-22:
Good to hear! I'll mark this issue closed.

Cordova itself is [already ported](http://cordova.apache.org/docs/en/5.0.0/guide_support_index.md.html) to Windows (8+, Phone 8+).

This plugin could also be ported. One would need to add C# implementations of all the functions declared in ble.js.

closed 2015-09-22 by fredrikeldh

RolfKH, 2015-09-22:
Hi Fredrik

Sounds good.

Do you have someone who can port the plugin for us please?

Regards,
Rolf

Rolf Kahrs Hansen, Dr.Scient
[logo_skjerm small]

Nattlandsveien 8, 5093 Bergen, Norway
eMail: rolf@spiromedical.com<mailto:rolf@spiromedical.com>
Telephone: + 44 20 81339064
Mobile: +47 9929 4592

Spiro Medical AS  is registered in Norway: NO 996 744 663 MVA

This e-mail and any files transmitted with it are confidential and intended solely for the use of the individual or entity to whom they are addressed. If you have received this e-mail in error or you are not the named addressee, please let us know by e-mail reply and delete it from your system. You should not disseminate, distribute or copy this e-mail, or disclose its contents to anyone. The recipient should check this e-mail and any attachments for the presence of viruses. Although the company has taken reasonable precautions to ensure no viruses are present in this e-mail, we cannot accept responsibility for any loss or damage arising from the use of this e-mail or attachments.

[http://www.mills-reeve.com/files/Uploads/Images/Mills%20&%20Reeve%20LinkedIn.png]<https://www.linkedin.com/company/3183464?trk=vsrp_companies_res_name&trkInfo=VSRPsearchId%3A2333039261410855463135%2CVSRPtargetId%3A3183464%2CVSRPcmpt%3Aprimary>