evothings/cordova-ble #77: Is there any documentation on how to use writeCharacteristic()?

u84six, 2015-09-21:
I see that the data argument asks for an ArrayBufferView, but I'm not sure how to set that up when trying to write:

format type: uint16 (2 octets)
value: 0x0001

How do I setup the ArrayBufferView to pass in that value?

fredrikeldh, 2015-09-21:
You could do this: ```new Uint16Array([0x0001])```.
Alternatively, this: ```new Uint8Array([0x00, 0x01])``` or ```[0x01, 0x00]``` to switch endian.

u84six, 2015-09-21:
I tried second solution because I need to switch endian, and I keep getting the same status 13 from onCharacteristWrite. Do you know what that status means? Even if I set the data param to some bogus string, I still get the same status code.

fredrikeldh, 2015-09-21:
On [Android](https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#CONNECTION_PRIORITY_BALANCED), 13 means  GATT_INVALID_ATTRIBUTE_LENGTH.

That's a very odd message for a characteristic; it indicates you shouldn't write 2 bytes to it.
I recognize 0x0001 though, perhaps you are meant to write to a descriptor, not a characteristic?

By the way, a general example can be found here: https://github.com/evothings/evothings-examples/blob/master/examples/arduino-ble/app/app.js

u84six, 2015-09-22:
In the spec for the device I'm working on, it says that for one characteristic, the format type is uint16/2 octets and can be any value from 0-2000. For the other characteristic is says it's a uint16/2 octets and should be set to 0x0001. I think ARM processor is big endian (although I did try both).

For the first characteristic, I'm passing:
var valueToWrite = new DataView(new ArrayBuffer(2));
                               valueToWrite.setUint16(0, 1, false);
and it's returning an error: 128

For the second characteristic, I'm passing the same thing (since they want a 1):
var valueToWrite = new DataView(new ArrayBuffer(2));
                               valueToWrite.setUint16(0, 1, false);
and it's returning an error: 13

I just wanted to check with you to see if this seems correct using your api.

Thanks
p.s. I did try your suggestions above but both were getting error: 13

fredrikeldh, 2015-09-22:
Yes, as far as I can tell, your code is correct.

u84six, 2015-09-25:
I'm still having trouble writing to a characteristic (error 13). I tried enabling the characterisitc for notifications before writing to it, but when I do, onCharacteristicChanged doesn't get called so I'm not getting the callback. enableNotifications is not throwing an error (i.e. gh.mGatt.setCharacteristicNotification is not returning false). Do you know what could be preventing onCharacteristicChanged from getting called?

fredrikeldh, 2015-09-25:
Writing to a characteristic will not, by itself, cause onCharacteristicChanged. That function only gets called when the remote device itself changes the characteristic's value.

Again I think the error 13 means you shouldn't be trying to write to that characteristic. Could you post a link to your device's specification? I feel that would be helpful in understanding its responses.

u84six, 2015-09-25:
Well if I look at the code in BLE.java, the enableNotifications method only calls the callback context on an error but not on success. I was under the impression that the gatt callback onCharacteristicChanged would be used to call the success callback. At any rate, when I call your enableNotification method, the success callback doesn't get called and there's no error. I don't see anything in logcat.

Unfortunately, is not extremely helpful. It just says that for a specific service, there's a characteristic with <uuid> that needs to be set to unint16/2 octets with value 0x0001. I've tried both big and little endian and no matter what I try, I get an error 13.

I'm going to contact the manufacturer and see if they can help.

fredrikeldh, 2015-09-25:
It is correct that enableNotifications doesn't call the success callback directly. That is because the function itself does very little except add an entry into the native callback table, and is such practically foolproof, so there's usually no need for feedback.

Have you tried writing 0x0001 to the descriptor instead of the characteristic yet?

u84six, 2015-09-25:
Ok thanks for the info.

Yes, I tried what you suggested in your first post and got error 13. At this point, I'm going to assume it's the device itself and try to contact the manufacturer. But thanks for all your help.

mikaelkindborg, 2016-09-02:
Closing this issue. Let me know if any issues persist, and I will reopen.

closed 2016-09-02 by mikaelkindborg