evothings/cordova-ble #79: Uncaught SyntaxError: Unexpected token ]

u84six, 2015-11-03:
I'm seeing a strange problem when calling readCharacteristic(). Every time I call it I get the error "Uncaught SyntaxError: Unexpected token ]" which I see in the browser (Chrome Inspector). There's no trace so I don't know how it's originating, but if I go into your code and modify onCharacteristicRead method and set mCurrentOpContext.success to pass it just a string, it works. But when it passes the byte array (which it should), I get this error. I'm using phonegap version 3.1.0. It's as if phonegap (or the Cordova code) isn't able to parse the byte array.

fredrikeldh, 2015-11-03:
An odd error. I don't remember seeing it before. I cannot reproduce it with Cordova 5 which we use normally.

I recommend you upgrade your Phonegap installation. Their latest version is 5.3.6.

u84six, 2015-11-10:
This problem can occur if you call: cordova.exec.setNativeToJsBridgeMode(cordova.exec.nativeToJsModes.LOAD_URL); prior to calling exec().

This fixed a different problem we had, but the side effect was that it also caused a problem when serializing byte arrays in the native-to-js bridge (i.e. calling a BLE method that returns a byte array). Just an FYI.

closed 2015-11-10 by u84six