evothings/cordova-ble #80: No error callback when trying to read characteristics if BLE is turned off

u84six, 2015-11-06:
When I make a call to BLE and the BLE device turns off, I'm not getting any kind of timeout and callbackContext.error() isn't ever called. It seems to hang in the Android BLE/GATT (example: if readCharacteristics is called, there's not timeout or error if it can't communicate with the device). How have you gotten around this?

fredrikeldh, 2015-11-06:
I'm unsure what you mean by "gotten around".

Our apps, such as they are, have a "Disconnect" button, which the user can press.

u84six, 2015-11-06:
I'm talking about making a request to BLE like readCharacteristic when the BLE device turns off. When that happens, readCharacteristic doesn't return an error if it can't communicate with the device, and seem to just hang at the Android SDK.

closed 2026-09-03 by u84six