evothings/cordova-ble #82: Trouble getting Cordova app working of an EvoThings Workbench example.

josephclawrence, 2015-11-19:
I am trying to create Cordova app, using PhoneGap, and initially using one of the RedBearLabs examples from EvoThingsWorkbench. I have followed all the instructions as far as I can see, and tried multiple different ways, also with Cordova CLI. I have added the Evothings BLE Cordova plugin, but it doesn't work.

I have added in some debugging code, and I can see that 'evothings.ble' is undefined. So it is at the point that evothings.ble.stopScan() is called that the app fails. 

I have tried to find where this evothings.ble object should be coming from but I can't find it - I'm not well versed enough in Objective C to dive in deeper, but according to the documentation I should be able to get this EvoThings Workbench example working as a Cordova app, with the correct plugins installed.

Thanks in advance for any help!

mikaelkindborg, 2015-12-08:
@josephclawrence Sorry for long response time!

I did a test build that worked, it would be great if you have the time to see if you can reproduce these steps.

Build machine: OS X 10.10.5, Xcode 7.1.1, Cordova 5.4.1

Phone: iOS 9.1 iPhone 4S

First I upgraded Cordova:

    sudo npm update -g cordova

Check Cordova version:

    cordova -v

Create test project:

    cordova create bletest com.evothings.bletest BLETest

Goto project folder:

    cd bletest/

Add BLE plugin:

    cordova plugin add https://github.com/evothings/cordova-ble.git

Add iOS platform:

    cordova platform add ios

Next, I took the files in the example app "BLE Scan" that comes with Evothings Studio Beta 2 download, and copied them to the "www" folder in the Cordova project.

The example app files and folders that should be copied to "www" are found here:

    EvothingsStudio_Mac_64_2.0.0_beta2_151202
        examples
            ble-scan
                app.js
                index.html
                libs
                ui

Then I built the app:

    cordova build ios

And finally opened the resulting Xcode project and deployed the app to the phone.

This worked for me. Let me know if there ios something different with your setup that might have caused the app to fail, missing JavaScript library files or something else.

josephclawrence, 2015-12-08:
Thanks for the reply Mikael.

I have actually done very similar to you a few days ago and it's working
100%. My original issue was specifically when using PhoneGap, which I still
never got to work, but now that I just compiled with Xcode rather, I just
gave up on PhoneGap.

mikaelkindborg, 2015-12-09:
@josephclawrence Great to hear that you found a solution, thanks for letting me know!

closed 2016-11-24 by mikaelkindborg