evothings/cordova-ble #88: How to get beacon data from ble.scan

MuraliKrishnaDev, 2016-02-29:
Hi,

I need to get beacon data from ble.scan i.e. UUID+Major+Minor+Txpower,

Please help me in this regard,

Thanks and Regards,
Murali

mikaelkindborg, 2016-03-21:
@MuraliKrishnaDev Hi, on iOS you cannot get iBeacon data using BLE scanning, you must use Apples CoreLocations API. Here is a plugin that does this for you: https://github.com/petermetz/cordova-plugin-ibeacon On Android, you can also use this plugin. (It is possible to use the BLE plugin for obtaining iBeacon data on Android, but the beacon plugin is a better option since it handles special cases and other things you would need to implement yourself.)

mikaelkindborg, 2016-03-22:
@MuraliKrishnaDev Closing this issue, please contact me if you want to reopen, or have any further questions, email mikael@evothings.com

closed 2016-03-22 by mikaelkindborg