evothings/cordova-ble #94: The link to convert the scanRecord isn't working anymore

angelorm, 2016-03-18:
The link in the ble.js file isn't working anymore and I really need to see how to convert the scanRecord to get the data (specially the manufacturer data). If you could fix the issue or leave a way to do it in here I would be thankful.

mikaelkindborg, 2016-03-21:
@angelorm Sorry for broken link. The library that provides parsing of the scan record on Android is here: 

https://github.com/evothings/evothings-libraries/blob/master/libs/evothings/easyble/easyble.js

We are planning to implement scan record parsing natively in the Android plugin.

mikaelkindborg, 2016-03-21:
Fixed broken link, available in this commit: https://github.com/mikaelkindborg/cordova-ble/commit/bde72d3b977dc5e6ae7df06d179291950d022fea

mikaelkindborg, 2016-03-22:
Merged update into master. Closing issue.

closed 2016-03-22 by mikaelkindborg