evothings/cordova-ble #97: writeCharacteristic problem

angelorm, 2016-04-14:
Hey everyone!

I'm trying to write a characteristic with an ArrayBuffer but I'm always getting the same value inserted whether is the value that I write. I've also used the nRF Master Control Panel app and tried to write it in there (as a ByteArray) and it worked, so the problem isn't related with the device.
This is what I'm doing to write it:
```
var ab = new ArrayBuffer(5);
var view = new Uint8Array(ab);
view[0] = 3;
view[1] = 2;
view[2] = 23;
view[3] = 76;
view[4] = 12;

 evothings.ble.writeCharacteristic(
             device.handle,
             char.handle,
             ab,
              function (win) {
                    console.log(win);
                     writeCharCallback.resolve(true);
              },
              function (errorCode) {
                     writeCharCallback.reject(errorCode);
              });
```

Am I doing something wrong? I probably am but I have no idea what.

nbezembinder1, 2016-04-14:
You seem to be writing an empty ArrayBuffer to your device instead of the Uint8Array.

angelorm, 2016-04-14:
I've changed the ab to view.buffer and the same thing happens. The weird thing is that it isn't writing just zeros. It show this values {158,233,101,0,0} and I have no idea where they come from.

nbezembinder1, 2016-04-14:
Could you try writing the Uint8Array directly using writeCharacteristic?

Like so:

```javascript
 evothings.ble.writeCharacteristic(
             device.handle,
             char.handle,
             new Uint8Array([3, 2, 23, 76, 12]),
              function (win) {
                    console.log(win);
                     writeCharCallback.resolve(true);
              },
              function (errorCode) {
                     writeCharCallback.reject(errorCode);
              }
);
```

angelorm, 2016-04-14:
Thanks for the help, it worked.
I thought I had to write the buffer instead of the Array, changed that and now it's working.

closed 2016-04-14 by angelorm