evothings/cordova-ble #98: iOS: peripheral.RSSI is deprecated

mikaelkindborg, 2016-04-28:
On iOS, peripheral.RSSI has been deprecated.

Updated file EVOBLE.m to use peripheral:didReadRSSI:error: instead.

scottmahr, 2016-07-05:
Is this actually fixed? I looked through EVOBLE.m but couldn't figure out if it had been changed or not.

mikaelkindborg, 2016-07-06:
I had a look and this has not been updated. peripheral.RSSI is used on line 489 in EVOBLE.m

Thanks for noticing!

scottmahr, 2016-07-06:
Does that mean that it will be fixed? Or is there a way I can fix it myself?

scottmahr, 2016-07-07:
> @scottmahr <https://github.com/scottmahr> You don't have to fix it, it
> should work even though it is deprecated (or does it not work?). Will look
> into updating this when working on a new release of the BLE plugin for iOS.

Yep, still works just fine. Thanks!

mdailor, 2017-02-07:
On iOS 10, when I try to read the RSSI of a scanned device:

evothings.ble.rssi(
    device,
    function (rssi) { console.log("RSSI=" + rssi },
    function (err) { console.log("ERROR=" + err });

I get the following error:

[NSTaggedPointerString count]: unrecognized selector sent to instance 0xa000000697373724
*** WebKit discarded an uncaught exception in the webView:decidePolicyForNavigationAction:request:frame:decisionListener: delegate: <NSInvalidArgumentException>
[NSTaggedPointerString count]: unrecognized selector sent to instance 0xa000000697373724

Is there a alternate method for reading the RSSI value? Thanks!

tozymandias, 2017-03-12:
@mdailor 

I am getting the exact same problem...in fact my app sometimes crashes due to this.  any luck fixing the problem?