evothings/cordova-ble #99: Build Failed Issue on Android

Rubinhuang9239, 2016-05-30:
I installed the ble plugin in my cordova project, now I have this build failed issue.
It's seems working with the plugin v1.1.2, but not the newest one?

Any body has an idea on this?
Thanks!

here is some reference logs:

```
A problem occurred configuring root project 'android'.
> Could not resolve all dependencies for configuration ':_debugCompile'.

   
> Could not find any version that matches com.android.support:support-v4:+.

     Searched in the following locations:

         https://repo1.maven.org/maven2/com/android/support/support-v4/maven-metadata.xml

         https://repo1.maven.org/maven2/com/android/support/support-v4/

         file:/Applications/android-sdk-macosx/extras/google/m2repository/com/android/support/support-v4/maven-metadata.xml
         file:/Applications/android-sdk-macosx/extras/google/m2repository/com/android/support/support-v4/
     Required by:
         :android:unspecified
```

mikaelkindborg, 2016-06-02:
@Rubinhuang9239 Hi, does it help to upgrade to Cordova 6? (Perhaps this is what you are using already?)

mikaelkindborg, 2016-07-10:
Does this problem persist? Closing issue for now, let me know how if to reopen.

Rubinhuang9239, 2016-07-21:
will try for sure, still using the old one. Thx!

mikaelkindborg, 2016-07-25:
@Rubinhuang9239 Hello, reopened the issue. Let me know if I can help in any way.

mikaelkindborg, 2016-11-24:
Closing issue due to inactivity, let me know if to reopen!

closed 2016-11-24 by mikaelkindborg