evothings/cordova-ble #100: readAllServiceData Setting issue

LAIZIHYIN, 2016-06-06:
I installed the ble plugin in my cordova project, now I have this build failed issue.
I use the connection, I want to use readAllServiceData function, but not executed and no error message, can I Setting it wrong?

my code
[image]

console.message
[image]

mikaelkindborg, 2016-06-22:
@LAIZIHYIN Hello, there might be an error when reading services/characteristics. Does it work to run any one the Evothings BLE example apps? Would be interesting to see if the EasyBLE library would work. I have not used the readAllServiceData function myself for a while.

To make it easier to debug an see where things go wrong, yoy could make a copy of the readAllServiceData function and rename it to eg myReadAllServiceData. the source code is here: https://github.com/evothings/cordova-ble/blob/master/ble.js#L672

Like this:

    function myReadAllServiceData(deviceHandle, win, fail) 
    // paste code here and insert log statements to trace execution

mikaelkindborg, 2016-10-12:
@LAIZIHYIN Hello, just want to say there is a new release of the BLE plugin, version 2.0.0, see release notes here: https://github.com/evothings/cordova-ble/releases/tag/2.0.0

This version has improved support for reading services and many more AIp updates and additions.

Did you get around to do further debugging of the read services problem?

mikaelkindborg, 2016-11-24:
Closing this issue due to inactivity, let me know if to reopen.

closed 2016-11-24 by mikaelkindborg