evothings/cordova-ble #104: Test multiple connections

mikaelkindborg, 2016-07-05:
Test that multiple BLE connections work (connecting to two BLE devices or more at the same time).

Suggest to use 'Dialog Semiconductor IoT Sensor' example app as a starting point, this app has been requested to be extended for connecting to two devices.

InventorSector, 2016-07-05:
Hello here too. First of all, i really enjoy u found that matter interesting! To be honest, i was expecting 'dialog semiconductors' to have a ready solution for multiple connections since it's 'internet of things'..
Anyway, i understand you don't have the modules to trial and error but seems you're java/iot expert!
I think that 3 things play major role here: 'device', 'address' and 'iotsensor'. 

In easyble.js there's 		var device = deviceObject;  //line 497
In app.js interesting lines seem to be:
 99-150 (	app.onScanButton = function()  )
app.onConnectButton = function(address)  )
	function addDeviceToView(device)
	function displayDeviceList()

mikaelkindborg, 2016-07-07:
@InventorSector Hi, I think that for the Dialog IoT Sensor library, you could try to create two instances of the sensor object. You should not have to change anything in easyble.js.

Here is a sketch of how app.js (this file https://github.com/evothings/evothings-examples/blob/master/examples/dialog-iotsensor/app/js/app.js) could be modified to work with two instances of the sensor object:

	var app = {};

	(function(){

		// Load script used by this file.
		evothings.loadScript('js/charts.js');
	
		var iotsensor1 = null;
		var iotsensor2 = null;
		var scanTime = 10;
		var devices = {};
	
		/* First function called, resets location to the scanning screen,			*/
		/* initializes the settings, creates an iotsensor.SFL object, starts		*/
		/*  scanning for IoT devices, and sets all callback functions.				*/
		app.initialize = function()
		{	
			initializeSettings();
		
			app.initializeSensor1()
		
			app.initializeSensor2()
		
			setEventHandlers();	
			initSensorFunctions();
			initSensorButtons();
		
			// Start scanning for iotsensor1.
			app.onScanButton();

			// TODO: When iotsensor1 has connected start scanning for iotsensor2.
		}
	
		app.initializeSensor1 = function()
		{	
			iotsensor1 = evothings.iotsensor.createInstance(
				evothings.iotsensor.SFL);
			
			iotsensor1
				.accelerometerCallback(ui.showSensorView)
				.gyroscopeCallback(ui.showSensorView)
				.magnetometerCallback(ui.showSensorView)
				.barometerCallback(ui.showSensorView)
				.temperatureCallback(ui.showSensorView)
				.humidityCallback(ui.showSensorView)
				.sflCallback(ui.showSensorView)
				.errorCallback(connectionError);
		}
	
		app.initializeSensor2 = function()
		{	
			iotsensor2 = evothings.iotsensor.createInstance(
				evothings.iotsensor.SFL);
			
			iotsensor2
				.accelerometerCallback(ui.showSensorView)
				.gyroscopeCallback(ui.showSensorView)
				.magnetometerCallback(ui.showSensorView)
				.barometerCallback(ui.showSensorView)
				.temperatureCallback(ui.showSensorView)
				.humidityCallback(ui.showSensorView)
				.sflCallback(ui.showSensorView)
				.errorCallback(connectionError);
		}
	...

The code in app.js would then have to be adapted to work with the two sensor instances iotsensor1 and iotsensor2.

This example is a bit special since it uses a high-level library for scanning and connecting to the Dialog Sensor Tags. Possible the library will have to be modified to work with multiple connections. 

Connecting to multiple BLE devices using easyble.js directly is another use case that also should be tested.

InventorSector, 2016-07-07:
Hey, thanx a lot.
I'm working on it trying to figure out how it could work.. still connects to only ONe.
I'm afraid that what we're searching here is the connection of device to address (i can read each module's address)and try to seperate them and connect to them like device1 and device 2, or simply have a global device term and try to connect to each address.. 
iotsensor.connectToDevice(
				devices[address]..

I think it means that from the 'devices' list, it connects the 'address' linked to the connect button that was pressed with the app.

Anyway i'm still trying to work on your way, but i'm afraid that maybe 'device' and 'address' could be the key..

nbezembinder1, 2016-07-08:
@InventorSector 

The `Dialog IoT Sensor App` is not designed to connect to multiple IoT Sensors, as mentioned on [Gitter.im](https://gitter.im/evothings/evothings?at=577feff77aeb0805278291b4).

You can build your own app using the `Dialog IoT Sensor Library` when following the[ starter-guide](https://evothings.com/dialog-iot-sensor-starter-guide/). You need two `iotsensor` objects as mentioned by @mikaelkindborg.

After initializing the objects you can use the `connectoClosestSensor` function in the library on each object. This function automatically connects to the closest IoT Sensor in the area. When using this function twice you can connect to each sensor like so:

```
	iotsensor1.connectToClosestSensor(
		7500, // Scan for 7500 ms
		function()
		{
			console.log("Connected to IoT Sensor 1");
		},
		function(error)
		{
			console.log('Disconnect error ' + error);
		}
	);
```

And for iotsensor2:

```
	iotsensor2.connectToClosestSensor(
		7500, // Scan for 7500 ms
		function()
		{
			console.log("Connected to IoT Sensor 2");
		},
		function(error)
		{
			console.log('Disconnect error ' + error);
		}
	);
```

However, this involves you creating your own application. Please follow the [starter-guide](https://evothings.com/dialog-iot-sensor-starter-guide/) and then try to expand it so you can connect to two IoT Sensors.

Also, this repo is not the right place for this issue. Maybe we can move it to https://github.com/evothings/evothings-libraries @mikaelkindborg ?

mikaelkindborg, 2016-07-08:
@InventorSector What I was thinking is that if you connect to the first IoT Sensor found, then wait a little, and connect to the second one found, than you are connected to both of them.

Now I just saw that @nbezembinder1 wrote an answer, thanks very much!

I guess you could also scan for devices yourself, in the app (not modifying the library code), and connect to devices.

Here is some "pseudocode" for that:

	var sensorOneIsConnected = false
	var sensorTwoIsConnected = false

	...

	evothings.easyble.startScan(
		function(device)
		{
			// Is it an IoT Sensor? (Here you can write your own version of 
			// isIoTSensor if you have specific requirements).
			if (isIoTSensor(device)) {
				if (!sensorOneIsConnected) {
					iotsensor1.connectToDevice(device, ...)
					sensorOneIsConnected = true
				}
				else if (!sensorTwoIsConnected) {
					iotsensor1.connectToDevice(device, ...)
					sensorTwoIsConnected = true
				}
			}
		},
		function(error)
		{
			...
		});

Does the above make sense?

mikaelkindborg, 2016-07-08:
@nbezembinder1 I think we will have to live with the issue in this repo, as far as I know there is no easy way to move issues. I guess I could create an issue in evothings-libraries or evothings-examples and point to this info, to make the connection clear.

nbezembinder1, 2016-07-08:
@mikaelkindborg 

Above is correct but I believe @InventorSector is only interested in reading and logging sensor data and this does not require the entire app. If you check the [starter-guide](https://evothings.com/dialog-iot-sensor-starter-guide/) on Evothings' website and scroll to the bottom you can see `app.js` for the guide which is only 58 lines of code, if you expand this you can easily add a second `iotsensor` object.

mikaelkindborg, 2016-07-08:
@nbezembinder1 Thanks a lot, great guide!

mikaelkindborg, 2016-07-10:
I am closing this issue since it does not really belong in the cordova-ble repository. Please open an issue in https://github.com/evothings/evothings-examples or https://github.com/evothings/evothings-libraries for further questions/issues related to the Dialog IoT Sensor library/example apps.

closed 2016-07-10 by mikaelkindborg

InventorSector, 2016-07-16:
@nbezembinder1   @mikaelkindborg
Guys sorry for being away, i was abroad for work and now i'm back!
I can see i've spiked your interest and of course many many thanx for your precious help!
It still seems really bizarre and frustrating that it's so difficult to connect more than 1 device in an IOT app..  I'm starting an issue in  https://github.com/evothings/evothings-examples  and I'm now trying to implement your proposals, thanx again!!!