evothings/cordova-ble #109: Empty String and/or odd characters from readCharacteristic

myyellowshoe, 2016-07-13:
Hey Fellas, and ladies :)

For some odd reason I'm getting an empty string and other seamingly random characters when I do a read on a characteristic. Running android with crosswalk on 4.3.

I've used nrfmastercontrolpanel ( https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) and things work great with a do a read on the characteristic. 

So far I've see "d", "2", "8", "5" "." "+"
I copied my logs and mashed them together not sure if they hold any significance. but somthing is definitely up.
%225.385.5225.2.858 2252.525.8285.5.235858+8583.+8.+5.522.29d2dd222+

Here's my code as well as some low level logs looking at the calls into readCharacteristic. Thoughts?

```
    module.readPower = function(){

        var characteristic = _findUuid(module.data.powerService.characteristics, "00002a19");
        var desHandle = (_findUuid(characteristic.descriptors, "00002902")).handle;

        var success = function(buffer){
          var data = evothings.ble.fromUtf8(buffer);
          console.log(data);
          /////// Returns "";
        };

        var failure = function(buffer){
          var data = new Uint8Array(buffer);
          console.log(data);
        };

        //Read data
        evothings.ble.readCharacteristic(module.data.deviceHandle, characteristic.handle, success, failure);
        return 0;
    };
```

`

/************* NRF CONTROL PANEL *************\
D/BluetoothGatt(13013): readCharacteristic() - uuid: 00002a19-0000-1000-8000-00805f9b34fb
D/BtGatt.btif(12866): btif_gattc_read_char
D/BtGatt.btif(12866): btgattc_handle_event: Event 1013
D/AbsListView(13013): unregisterIRListener() is called
D/BtGatt.btif(12866): btif_gattc_upstreams_evt: Event 3
I/BtGatt.btif(12866): set_read_value unformat.len = 1
D/BluetoothGatt(13013): onCharacteristicRead() - Device=19:00:00:00:00:19 UUID=00002a19-0000-1000-8000-00805f9b34fb Status=0
D/AbsListView(13013): unregisterIRListener() is called
I/AudioService(  331): getStreamVolume 3 index 40
D/AbsListView(13013): unregisterIRListener() is called
D/AbsListView(13013): unregisterIRListener() is called
E/Watchdog(  331): !@Sync 1351
D/STATUSBAR-NetworkController(  807): refreshSignalCluster: data=-1 bt=false
D/STATUSBAR-IconMerger(  807): checkOverflow(192), More:true, Req:true Child:16
D/lights  (  331): button : 0 +
D/lights  (  331): button : 0 -
I/AudioService(  331): getStreamVolume 3 index 40
E/SMD     (  161): DCD ON

/************* CORDOVA APP *************\
I/Sensors (  331): Enabled 9 axis sensor fusion
I/MPL-mldl_cfg_mpu:(  331): inv_mpu_resume(,,,,0000) -> 0000
V/Sensors (  331): Stopping DMP
I/Sensors (  331): Disabled 9 axis sensor fusion
I/MPL-mldl_cfg_mpu:(  331): inv_mpu_suspend(,,,,7fff) -> 0000
D/BluetoothGatt(13639): readCharacteristic() - uuid: 00002a19-0000-1000-8000-00805f9b34fb
D/BtGatt.btif(12866): btif_gattc_read_char
D/BtGatt.btif(12866): btgattc_handle_event: Event 1013
I/AudioService(  331): getStreamVolume 3 index 40
D/BtGatt.btif(12866): btif_gattc_upstreams_evt: Event 3
I/BtGatt.btif(12866): set_read_value unformat.len = 1
D/BluetoothGatt(13639): onCharacteristicRead() - Device=19:00:00:00:00:19 UUID=00002a19-0000-1000-8000-00805f9b34fb Status=0
D/STATUSBAR-NetworkController(  807): refreshSignalCluster: data=-1 bt=false
D/STATUSBAR-IconMerger(  807): checkOverflow(192), More:true, Req:true Child:16
I/AudioService(  331): getStreamVolume 3 index 40
I/chromium(13639): [INFO:CONSOLE(624)] "No Content-Security-Policy meta tag found. Please add one when using the cordova-plugin-whitelist plugin.", source: file:///android_asset/www/js/jslog.js (624)
I/AudioService(  331): getStreamVolume 3 index 40
D/STATUSBAR-NetworkController(  807): refreshSignalCluster: data=-1 bt=false
D/STATUSBAR-IconMerger(  807): checkOverflow(192), More:true, Req:true Child:16
E/SMD     (  161): DCD ON`

myyellowshoe, 2016-07-13:
Ok i've figured out a bit more. It looks like the byte array is getting returned properly from the device. In my case i'm looking for a number returned from the device.

I debugged the actual function running on the java side.

```
public void onCharacteristicRead(BluetoothGatt g, BluetoothGattCharacteristic c, int status)
		{
			Log.d("DEBUG", "#~#~#~# characteristic read: " + c.getValue()[0]);
			if (status == BluetoothGatt.GATT_SUCCESS) {
				mCurrentOpContext.success(c.getValue());
			} else {
				mCurrentOpContext.error(status);
			}
			mCurrentOpContext = null;
			process();
		}
``` 
Gives me:
```
#~#~#~# characteristic read: 100
```

Just what I need. So in since this is the proper value, its a problem with how I'm reading the buffer in javascript.

Shouldn't the following be more than needed?

```
          var data = evothings.ble.fromUtf8(buffer);
          console.log(data);
```

mikaelkindborg, 2016-07-13:
@myyellowshoe Hello, what is the format of the data in the buffer? Is it a utf8 string?

myyellowshoe, 2016-07-13:
@mikaelkindborg It should just be a 1 byte array. Not sure on the specifics of the format. I just know it returns a number.

myyellowshoe, 2016-07-13:
Woot! So as I figured it was all about handling the buffer in the proper way. My poor understanding of buffers was the culprit.

In my case doing the following got me the value I needed. 
```
(new Int8Array(buffer)).toString();
```

closed 2016-07-13 by myyellowshoe

mikaelkindborg, 2016-07-15:
@myyellowshoe Great to hear you solved the problem!