evothings/cordova-ble #110: Read/Write after connecting problem, password issue?

aserrin55, 2016-07-15:
I have scanned devices and everything is OK.
But when I try to connect with any device there is no pin requirement like:

[image]

I'm sure that the device is password protected, but the input dialog does not appear.
So, I make the connection and the status is connected, but when I want to read something, there are some characteristics encrypted and its value is not readable. But the "free" characteristics can be read properly. For example:

[image] 

What should i do? Where is the password need to be included? My connection code:

     $scope.connect = function(addressParam) {
      var params = {address: addressParam};

      new Promise(function (resolve, reject) {
        bluetoothle.connect(resolve, reject, params);
      }).then(connectSuccess, handleErrorC);

      function connectSuccess(result) {
        console.log("connectWithExit");
        console.log(result);
      }

      function handleErrorC(result) {
        console.log("ERRORalConectarse");
        console.log(result);
      }
    }

Thanks

mikaelkindborg, 2016-07-15:
@aserrin55 Hi, this is likely because the plugin does not support password pairing and encrypted connections. Looks from the screenshots you tested this on Android. If you have tested on iOS, did you get the same result?

aserrin55, 2016-07-15:
Don't know, i'm not able to make the test on iOS. How can I do this in order to read properly every characteristic? Do you have any idea? Thanks again.

mikaelkindborg, 2016-07-19:
@aserrin55 My understanding is that the BLE plugin needs to be updated to support secure connections. I cannot say when this could happen. If you wish to implement it you are most welcome of course. You could also have a look at the alternative BLE plugins and see if they support the functionality you need: 
* https://github.com/don/cordova-plugin-ble-central 
* https://github.com/randdusing/cordova-plugin-bluetoothle

aserrin55, 2016-07-20:
I have made lots of proofs with that plugins too, without results. I will implement the necessary functionality, will take long hahah. Any guide, resource, or advice to make a middle layout between Angular and Java?

Thank you so much for all your attention @mikaelkindborg

mikaelkindborg, 2016-07-25:
@aserrin55 Hello again, the secure paring would have to be implemented natively, in Java for Android. If you modify the Evothings BLE plugin, there may need to be added new functions in JavaScript. I am not familiar with secure paring/bonding. What kind of BLE device do you have? I could try to get my hands on a BLE device that has secure characteristics and read up on the subject to help with this. Do you have any pointers/links to tutorials/references that outline what you want do do?

aserrin55, 2016-07-26:
I have a beacon device. The unique reference that I have is the official SDK of the gadget. Here it is: https://github.com/jaalee/Jaalee-Beacon-Android-SDK-master
Thanks

mikaelkindborg, 2016-07-29:
@aserrin55 Thanks will try to find some device to test with.

mikaelkindborg, 2016-10-12:
@aserrin55 Some months later there is now support for pairing/bonding in the plugin. See release notes for details: https://github.com/evothings/cordova-ble/releases/tag/2.0.0

Regarding the problem that the paring dialog is not shown on Android, I have experienced that too. There is a new function now, evothings.ble.bond() which is used to initiate pairing and show the dialog. See above release notes for details. However, I have noticed that sometimes paring fails, and you have to call bond() again. My impressing is that pairing/bonding is not yet fully mature, not many BLE devices use this to my knowledge.

For all who wants to test, here is a new APK with Evothings Viewer for Android with the latest BLE plugin release 2.0.0: https://evothings.com/uploads/evothings-studio/viewer/EvothingsViewer_1.5.0_161012_1430.apk

mikaelkindborg, 2016-10-12:
Related issue: https://github.com/evothings/cordova-ble/issues/122

mikaelkindborg, 2016-11-24:
Closing this issue, let me know if there are any recent activity and if to reopen.

closed 2016-11-24 by mikaelkindborg