evothings/cordova-ble #111: connect to devices TOO SLOW -- up to 80 seconds or more

19317362, 2016-08-12:
How can I deal with this?

mikaelkindborg, 2016-08-12:
@19317362 80 seconds sounds pretty bad. I have experienced 10-20 seconds on some devices. 

It can help to read only the services you need. There is a description of this in this tutorial: https://evothings.com/doc/tutorials/how-to-connect-to-ble-devices.html Search for "When reading services, you can specify the services to read (this can improve the performance):" on the page to find the relevant section.

It might be that some Android phones (older ones in particular) and/or BLE devices are slower than usual. Have you had a chance to test with different BLE devices/phones to compare?

If there is a native Android app available for your device, you can test how long it takes to connect.

What BLE device are you using? I can try to find it and do some tests.

mikaelkindborg, 2016-09-02:
@19317362 Is it possible to provide further details on how to reproduce the problem? What BLE device you are using and what Android phone/OS version used would be good to know. If you have code you can share that would also be great.

mikaelkindborg, 2016-11-24:
Closing issue due to inactivity, let me know if to reopen.

closed 2016-11-24 by mikaelkindborg