evothings/cordova-ble #113: EasyBLE library gives error: EASYBLE_ERROR_CHARACTERISTIC_NOT_FOUND

mikaelkindborg, 2016-08-31:
(Putting this issue in the cordova-ble repo since it might be related to the underlying plugin code, and to keep BLE issues at one place.)

Developer reported calling writeServiceCharacteristicWithoutResponse() and getting error message:

    LOG: Error: EASYBLE_ERROR_CHARACTERISTIC_NOT_FOUND 00001802-0000-1000-8000-00805f9b34fb:00002a06-0000-1000-8000-00805f9b34fb

This happened when using a Cypress BLE card.

Attaching file with test app that was used when this error occurred. There have also been reports of EASYBLE_ERROR_CHARACTERISTIC_NOT_FOUND when using the BBC microbit.

[Cypress-FindMe.zip](https://github.com/evothings/cordova-ble/files/446740/Cypress-FindMe.zip)

storm1er, 2017-02-04:
Getting the same error using bleno on RPi 3 : https://github.com/sandeepmistry/bleno/issues/269

Bluetooth LE Scanner on android says ok : https://goo.gl/photos/i3LujZuW2riMNBZc8

storm1er, 2017-02-04:
ok, uuids must be in "ffffffff-ffff-ffff-ffff-fffffffffff1", and no "fffffffffffffffffffffffffffffff1"