evothings/cordova-ble #119: App keeps crashing on iOS 10

qpwr, 2016-09-18:
Hi @mikaelkindborg,

I'm experiencing issues on iOS 10. Scanning and connecting works, but the app crashes to background immediately after connecting. I don't have any error messages because I yet don't have XCode 8 installed, but just want to ask if you can confirm your plugin and the methods are working correctly on iOS 10. I will try to provide more information ASAP.

Marc

mrcbnsls, 2016-09-19:
Just had the opportunity to test this with XCode 8. This is the actual error message:

```
Assertion failed: (![self.commands isEmpty]), function -[EVOPeripheral assertCommandAvailable], file /Users/qpwr/CordovaApps/SmartRace/platforms/ios/SmartRace/Plugins/cordova-plugin-ble/EVOBLE.m, line 289.
```

qpwr, 2016-09-19:
Would be great if you'd be able to shed some light on this very soon, because the app is currently not usable on iOS 10 :-(

qpwr, 2016-09-21:
@mikaelkindborg, do you already have an idea whats wrong? Do you think there is a chance you could provide me with some sort of quickfix?

mikaelkindborg, 2016-09-22:
@qpwr Hi, sorry for slow response, have been upgrading iOS and Xcode now and done testing. Here is a test report and an analysis.

### Test result

Tested BLE on iOS 10.0.1 on iPad Mini.

Could no reproduce the problem you found. Tested various of the Evothings BLE example apps using different models of the TI SensorTag.

### Possible causes

The assertion happens when a native callback (delegate method) in iOS occurs, but no corresponding JavaScript callback (Cordova command) exists. This is a bit nasty, it would be fault-tolerant to use conditionals and only call the JavaScript callback when a command exists.

I can think of two causes for this happening:

* iOS 10 has some behaviour causing a delegate method to be called, or be called multiple times, even though no corresponing request has be made (no JavaScript function have been called to trigger the delegate method callback).

* The app has some behaviour that causes this to happen (not seen this previously however).

### Further test data

Do you have any details on which JavaScript function (in the BLE plugin API) is causing this? Would it be possible for you to log what is happening up to the assertion occurs? That would be helpful in finding the problem. Also details on which BLE device you are using could be helpful.

### Possible workaround

The workaround I can think of is to replace the method call to assertCommandAvailable with a conditional, in the file EVOBLE.m. That is 7 occurrences.

Add this method:

    - (bool) noCommandIsAvailable
    {
        return [self.commands isEmpty];
    }

Then replace this (7 places):

    [self assertCommandAvailable];

With this:

    if ([self noCommandIsAvailable]) return;

Not sure how this would work out, but is worth trying.

### Related issues

Possibly related: 

https://github.com/don/cordova-plugin-ble-central/issues/311
https://forums.developer.apple.com/thread/63704

mrcbnsls, 2016-09-22:
Hi @mikaelkindborg, thanks A LOT for your response. The workaround you proposed works great and I will hopefully be able to release a new version of my app tonight after some more testing. I will try to dive a bit deeper into the issue and what you mentioned as possible causes afterwards and try to give you some more feedback on it. Big thanks so far!

mikaelkindborg, 2016-09-22:
@qpwr Great to hear it works! Perhaps we should put this fix into the plugin production code? Any hints you may have regarding what causes the situation with the assert would be useful for finding out what goes wrong. Best of luck with your app!

mikaelkindborg, 2016-10-10:
What is the status of this issue? Should we put the fix into the plugin? Anyone seeing crashes on iOS 10?

sjrcgtek, 2016-10-10:
For what its worth, my app works fine on iOS 10.  I scan for and connect to
devices, read services, read/write characteristics and register for (and
receive) indications all exactly as I did before.

qpwr, 2016-10-10:
Well, I can only say that the app was reproducably crashing on iOS 10 without the fix and does now no longer. I still don't know the exact reason, but I know its fixed now :-D - so I would vote for adding the fix to the plugin. Thanks!

mikaelkindborg, 2016-10-12:
@sjrcgtek @qpwr Many thanks for the input! Will have a look at adding the fix on iOS.

mikaelkindborg, 2016-10-12:
Fix added in this commit: https://github.com/evothings/cordova-ble/commit/074e0a8de77f25ccb145211c77ffe3f270022724

closed 2016-10-12 by mikaelkindborg