evothings/cordova-ble #121: Samsung S4 (Android 5.0.1) Re-Connexion Problem

meddev4, 2016-10-04:
Hi Team, 

We are developing an IoT Gadget, we used Evothings (Latest .JS Files and Viewer)+jQuery Mobile+PhoneGap. We build an .APK for testing on Android smpatphones. Connecting to our product for the first time is OK, the second try (re-launchin the app) we lose connection with the IoT gadget. This problem is senn only on Galaxy S4 and Ace 3. On iPhone 5, Galaxy S5, Galaxy S6 Edge it's OK, no problem to reconnect again and again...

- S4 = Android 5.0.1
- S5 = Android 5.0.2
- S6 = Android 6.0.1

The issue is seen on Evotnings Viewer (Old and newer version) and on the .APK (Build using PhoneGap).

The steps are very simples:

1. We connect the first time from our phone to the BLE produtct we are developing

2. Communication is OK, data is sent and  received 
3. We close the application from the smartphone (Without turning off Bluetooth/BLE from smartphone, so the phone is always paired to our BLE product).
4. We relaunch the application---> No data is received or sent to the BLE product.

Best regards,
Mohamed

mikaelkindborg, 2016-10-04:
@meddev4 Thanks for reporting this issue!

meddev4, 2016-10-06:
Hi @mikaelkindborg,

We don't list the device programmatically from our Evothings application.

The device is listed and paired manually from phone native settings screen, then we connect to it's BLE characteristic from our application.

It's difficult to post some logs because the application is packaged in an .APK

mikaelkindborg, 2016-10-10:
@meddev4 Got an S4 now and done testing.

The app I have been testing with is here:

https://github.com/mikaelkindborg/cordova-ble/tree/master/examples/core-api/hexiwear-bonding

The test app uses the HexiWear device. It runs on the S4, but with some errors. In particular, there is error 133 on connect. When this happens the app waits a little, then connects again and after a few attempts the connection succeeds. The native HexiWear app was not able to connect at all on the S4 when I tested (displaying error 133 in logcat).

My understanding from the testing and reading I have done is that error 133 can be avoided or made appear less frequently by waiting a little before connecting, and also to reconnect again if error 133 appears. Check file app.js for code you can use to experiment, like this code: 

https://github.com/mikaelkindborg/cordova-ble/blob/master/examples/core-api/hexiwear-bonding/app.js#L252

Tested the app also on LG G4, works much more stable but still error 133 happens on connect now and then.

I noted that sometimes bonding fails in the app. You then have to bond again (press Connect button again). The app does explicit bonding. If you bond using Settings, the app should detect that the device is already bonded and connect directly.

I have also updated the Android BLE plugin to make BLE calls on the UI thread. There are also new calls for bonding/paring. The latest code for the BLE plugin is here:

https://github.com/mikaelkindborg/cordova-ble

You should be able to install this plugin version by using the command:

    cordova plugin add https://github.com/mikaelkindborg/cordova-ble

I did the testing using an updated version of Evothings Viewer, which is available here:

https://evothings.com/uploads/evothings-studio/viewer/EvothingsViewer_1.5.0_161010_1515.apk

Let me know how this goes. I very much would like this to work as well as possible.

Best regards, Mikael

mikaelkindborg, 2016-10-12:
Here is a new APK for Android with the latest BLE plugin release 2.0.0: https://evothings.com/uploads/evothings-studio/viewer/EvothingsViewer_1.5.0_161012_1430.apk

mikaelkindborg, 2016-10-14:
@meddev4 Any news on the S4 issues?

mikaelkindborg, 2016-11-24:
Closing this issue due to inactivity, let me know if to reopen.

hoongoon86, 2017-01-12:
Samsung Galaxy S4 with Android 5.0.1. 
Using connection structure as, 

- Check bonded device. (getBondedDevices)
- if no bonded device or the device is not what I'm looking for, start scanning. (startScan)
- if bonded device is found, start connecting. (connectToDevice)
- Then do the rest. Like getting services, characteristics, reading, writing, notification, etc. 
- If there's no device around while scanning, then 'No device found' message appears. 

Android 6 is fine. (Tested with Nexus 5)
Press connection button in the app then it does what it suppose to do. 
However in S4, it never connects when connection button is pressed. (Not sure if its never or ever. I only tested for 30~40 times)

To make it work, I had to goto the bluetooth settings, find the device then click to pair, (usually pairing fails with error message, so try couple more times). 
Once it's paired, then open the app and click to connection button. Then BOOM! Connected. Nice and fast. 

Not really like the way it is. Hope we can have other solution to make it work.

mikaelkindborg, 2017-01-25:
@hoongoon86 Hello, and sorry for slow response. From what I can tell your connect flow looks fine. One trick I used was to call startScan repeatedly if connectToDevice fails, with a setTimeout that started scan after a second or so (did this both if it failed with bonded device that was found, and when connecting from startScan failed). I also noted that this "trick" was used with the HexiWear native Android app. I think it was on an S5 that I observed cases when the native app failed and the Cordova app succeeded in connecting, interestingly. My guess is that bonding is not fully mature on Android, in particular earlier OS versions. I have not used the S4. Perhaps you will have to resort to supporting only Android 6 and higher. There can also be device specific problems, like the S4 might have problems other phones with Android 5 do not have. It can be very hard to tell.