evothings/cordova-ble #123: iOS - UIBackgroundModes: State restoration of CBCentralManager

valentiniljaz, 2016-10-06:
On iOS 9.3+, if I remove UIBackgroundModes (bluetooth-central) key from Info.plist, app failes to successfully scan for devices while in foreground. If I add the above key back into Info.plist the app finds devices. Is there any way to use the plugin (in foreground) without specifying UIBackgroundModes in Info.plist file?

When scanning in foreground (without UIBackgroundModes specified) I get an error: State restoration of CBCentralManager is only allowed for applications that have specified the "bluetooth-central" background mode.

mikaelkindborg, 2016-10-10:
@valentiniljaz Hi, I think the error comes from methods being called in the plugin that requires background mode. Ideally there should be a config for this (like a parameter to the Cordova command for adding the plugin).

As a patch you can try to comment out the code on this line: https://github.com/evothings/cordova-ble/blob/master/src/ios/EVOBLE.m#L1297

Like this:

    self.central = [[CBCentralManager alloc]
        initWithDelegate: self
        queue: nil
        options: @{
            //CBCentralManagerOptionRestoreIdentifierKey: @"EVOCentralManagerIdentifier",
            CBCentralManagerOptionShowPowerAlertKey: @YES }];

Let me know how it goes.

mikaelkindborg, 2016-10-10:
Here is a discussion of specifying variables when adding a plugin: http://stackoverflow.com/questions/29645780/settings-variable-in-cordova-config-xml-for-plugin-installation

We could use this method to config background support, then there would be no need of patching the source code.

valentiniljaz, 2016-10-12:
Thanks for your help. Commenting out the line you mentioned did the trick.

closed 2016-10-12 by valentiniljaz

mikaelkindborg, 2016-10-14:
@valentiniljaz Great that it worked! Will find a way to configure this. Opened a new issue for making a configure option: https://github.com/evothings/cordova-ble/issues/127