evothings/cordova-ble #128: startScan() does nothing on tablets

PTthe13, 2016-10-25:
I'm having a problem with two tablets and the startScan() _(in **ble** and in **eddystone**)_ function.

I'm trying to implement scan on 2 tablets. 
Two Asus. One TF300 with Cyanogenmod and Android 5.1.1 and another, the Asus K013 with stock Android 4.4.2.

In either of the cases, the startScan() doesn't return anything (success or error).

It works fine on every other device we've tested. From iPhone 5 and 6 _(both **s** and **"normal"**)_ , Oneplus Two, and several other low-end devices.

Is there anything I can do to make the scan work, or at least return something?

Thanks in advance

mikaelkindborg, 2016-10-26:
@PTthe13 If you have adb logcat installed there may be hints in the log about what goes wrong. The best is if it is possible to debug. Do other BLE apps that scan work? If they work this indicates a bug in the BLE plugin. There could be tweaks that are needed to make BLE work on Cyanogenmod. Android 4.4.2 has early BLE support and may have problems, but I have run several BLE apps on 4.4.2 (my main problem was that the BLE cache was filled up so that BLE stopped all the time).

PTthe13, 2016-10-26:
@mikaelkindborg The Asus K013 had an application before that did scan BLE beacons. But it worked very poorly. "Very" being an understatement. 

Is there a way to easily clear the BLE cache so I can try that out?

mikaelkindborg, 2016-10-27:
@PTthe13 For clearing the BLE cache on Android you can try this app by Radius Networks: https://play.google.com/store/apps/details?id=com.radiusnetworks.bluetoothcrashresolver

Related blogpost: http://developer.radiusnetworks.com/2014/04/02/a-solution-for-android-bluetooth-crashes.html

Another more radical solution is to factory reset the tablet.

If above methods fails, one needs to root the tablet and manually remove entries in some BLE settings file (don't have the details at hand unfortunately). There is info on how to do this I noted this when I googled for the problem, but I did not save the link!

Let me know how it goes.

mikaelkindborg, 2016-11-24:
@PTthe13 Any progress on this issue?

PTthe13, 2016-11-24:
@mikaelkindborg I haven't been able to test the latest info you gave me. Other projects got in the way, but I will return to the project in question and give you feedback as soon as I can.

Thanks.

PTthe13, 2016-12-16:
@mikaelkindborg just got back to this project and to try to fix this. Tried with the app you mentioned. No luck there.

Unfortunately I can't factory reset the tablet or root it because it is from our client, it's not our device.

I'll keep looking into this. If I find another solution I'll post it here :)

PTthe13, 2016-12-16:
Forgot to mention. Logcat "seems" to find beacons. Only the plugin "doesn't return" anything (If that makes any sense).

Example of logcat:
`D/BtGatt.btif( 8129): btif_gattc_upstreams_evt: Event 4096
D/BtGatt.GattService( 8129): onScanResult() - address=C1:F1:CF:D2:34:C7, rssi=-81
I/bt-hci  ( 8129): BLE HCI(id=62) event = 0x02)
I/bt-hci  ( 8129): btu_ble_process_adv_pkt
D/BtGatt.btif( 8129): btif_gattc_upstreams_evt: Event 4096
D/BtGatt.GattService( 8129): onScanResult() - address=38:01:95:CA:88:B8, rssi=-65
I/bt-hci  ( 8129): BLE HCI(id=62) event = 0x02)
I/bt-hci  ( 8129): btu_ble_process_adv_pkt
D/BtGatt.btif( 8129): btif_gattc_upstreams_evt: Event 4096
D/BtGatt.GattService( 8129): onScanResult() - address=EA:FC:8D:5C:23:D8, rssi=-66`

But when I go to the console in Android Studio _(should return this: 'Scanned for beacon...' when a beacon is found)_, nothing...

mikaelkindborg, 2016-12-20:
Does this mean that you now get scan results in JavaScript?

PTthe13, 2016-12-20:
@mikaelkindborg no. In javascript there is nothing. I recently "made an alternative" version of the app with the https://github.com/petermetz/cordova-plugin-ibeacon and it works _(iBeacon protocol...)_. It is the "workaround" I'm using at the moment cus of the damn tablets not working.

In my particular case it will suffice. But there is something "missing". The information is not going from the "logcat" to the "javascript" in the tablets. :\

PTthe13, 2017-11-23:
Any news on this? Just tried on another tablet (a similar project) and it doesn't work.

It displays the exact same problems as before.
Logcat shows:

> D/BtGatt.GattService( 4382): onScanResult() - address=XX:XX:XX:XX:XX, rssi=-58
> D/BtGatt.btif( 4382): btif_gattc_upstreams_evt: Event 4096
> D/BtGatt.GattService( 4382): onScanResult() - address=XX:XX:XX:XX:XX, rssi=-87
> D/BtGatt.btif( 4382): btif_gattc_upstreams_evt: Event 4096
> D/BtGatt.GattService( 4382): onScanResult() - address=XX:XX:XX:XX:XX, rssi=-73
> D/BtGatt.btif( 4382): btif_gattc_upstreams_evt: Event 4096

But nothing is reported on the javascript part

The tablet is a Samsung Galaxy E (SM-T560) with android 4.4.4 (latest one available).