evothings/cordova-ble #130: Background mode every whatever minute

hoongoon86, 2016-11-25:
Hi, 

I know it's not an issue of this plugin. 
Actually, this plugin works great on iOS background mode.(Tested on iPhone 5, iOS 9.3) 
When the app goes on background, BLE can scan, connect, and read / write. 

What I want to do is to run BLE on background every ~ minutes. (like 5m, 10m, etc)
Since I'm using cordova / phonegap, I was trying to implement simple javascript timer and local notification but none of these can wake up the app and run BLE. 

Does anyone have an idea how to make this happen?  
Thank you!

hoongoon86, 2016-12-01:
Update:

Want to share a bit of trick here. 
For iOS, there are mainly 3 activities of BLE that keeps BG alive. Connecting, read and write. 
We should be more careful with 'Connecting' function. 
If interested, please refer 
[https://developer.apple.com/library/content/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html](url)

But the issue I'm having was slightly different. 
Calling `connectPeripheral:options` can keep BG alive until connected. 
What happens after connected? Well, BG is ended and app is moved to suspended state. 
So, I implemented this but I still need more methods to resolve my issue. 
On BG, once it's connected, I started to call `writeCharacteristicWithoutResponse`. 
I've noticed that this keeps BG alive for 4~5 seconds. 
Now, I'm keep calling this method every 4 seconds.

mikaelkindborg, 2016-12-06:
@hoongoon86 Hi, thanks for sharing this, sorry for not being of much help here. I think BLE scanning will work in the background, but I don't think you can control how often the scan is performed/the app wakes up. Perhaps there are other background tasks/services that could be added to the app to wake the app up?

hoongoon86, 2016-12-06:
@mikaelkindborg Hi, thanks for your comment. 
Unfortunately, I've failed to keep background alive using scanning function. I was trying to scan for non-existing services and trying to end scanning when the app comes back on foreground. Perhaps I was doing it wrong or simply it's not how it's supposed to be. I don't know. 
I couldn't find other solution (methods or functions) to make BLE work on background. 
Please share us if you find any. 

So far, I could make BG work for about 36 hours. Of course it consumes energy from both phone and the device but it was a good experiment.

mikaelkindborg, 2016-12-06:
@hoongoon86 I will keep my eyes open for solutions and ideas!