evothings/cordova-ble #132: Not able to receive data as an ArrayBuffer in peripheral mode....

rajatsawant, 2016-12-28:
[Ionic code.txt](https://github.com/evothings/cordova-ble/files/675349/New.Text.Document.txt)

The data I receive is in string format and a garbage value, changing every time i send the same message from a central ble.

mikaelkindborg, 2016-12-29:
@rajatsawant Hi, peripheral mode is still largely untested. What happens if you send a very simple value, like a byte with value 1 for example? Does that arrive as expected? Is it an array of bytes that does not work?

rajatsawant, 2016-12-30:
Thanks for replying.
In the onWriteRequest callback, i am supposed to get an ArrayBuffer as the data from the master. But I get a garbage string. And it changes every time even though the data from the master is the same.
Regards.

mikaelkindborg, 2016-12-30:
@rajatsawant This sounds like a bug in the plugin. Not sure when we will have time to address this however.