evothings/cordova-ble #136: Can't disconnect BLE devices when killing the app with background plugin on

Albertazzi88, 2017-03-07:
I'm facing an issue not directly related (I guess) to this plugin, but since I suspect it is a conflict or a strange behaviour in the bluetooth connection, I guess if someone who better knows how to manage BLE devices can help me understand how to solve this issue.

I'm developing an app that interfaces with BLE devices.
Inside the app I can pair the devices and monitor their status.
If I close the app I want to disconnect the devices from the smartphone, and when I open it again, I want the app to reconnect automatically to known devices.

Everything works fine without using the [background plugin](https://github.com/katzer/cordova-plugin-background-mode), but I need to use it since the app needs to monitor the BLE devices even when it's in background.

The problem happens ONLY if I use the background plugin: if I open again the app, the known devices are not reconnected INSIDE the app, but it appears they are already/still connected to the smartphone from the previous time (even if not shown in the bluetooth devices in the phone settings).
It seems that the bluetooth connection to known devices isn't really being closed when I kill the app.
It appears like the device is still connected as therefore it doesn't show up when I scan in the app.

I can't reconnect until I close the app, switch off and then on again bluetooth, and reopen the app.

The steps are very simples:

1)    I open and connect the first time from the phone to the BLE product
2)    Communication is OK, data is sent and received
3)    I close (kill) the application from the smartphone (Without turning off Bluetooth/BLE from smartphone)
4)    I relaunch the application ---> Known device is not reconnected, but appears to be somehow still paired with the phone (if I press the pairing button it does not go into pairing mode, that happens if it is already bonded to a phone).

Why this behaviour? Looks like something in the background plugin prevents the bluetooth connection to be closed on app closing. I specify that:

-    it only occurs using the background plugin, without it activated everything works ok
-    this behaviour occurs only with Android (don't use the background plugin for iOS)
-    it does happen only on some Android smartphones: tried three different phones, with Adroid version v4.4, v5.1.1 and v6.0.1, and only two of them show this issue (android version 4.4 on S3 and 6.0.1 on S5 Neo).

Thanks

Albertazzi88, 2017-03-10:
After some testings, it appears that it is an issue caused by the Foregroundservice which doesn't close when I close the app.
I found out that forcing the closure of the app in the settings solve the problem, and when I open again the app everything works. I believe then that it is a problem of a service (probably the one created by the background plugin), which is somehow not stopped when I swipe away the app.

I tried to add the stopWithTask = "true" in the Android Manifest in the service field (both for the background service and for the beacon/bluetooth service, terminating one of them would cause the disconnection of the ble devices from the phone), but this doesn't force the service to stop when I swipe away the app! How can this happen? Do you have any suggestion about that, or things I must be aware of when trying to close a service together with the app?
I also tried to add "onTaksRemoved" and call "stopService" inside, so that the service should be stopped when the activity is closed, but this does not work too.

Any idea why it behaves like this? How can I force the service created by this plugin to stop?

Can it be that the some service is not "binded" to the app, and therefore it doesn't see the stopWithTask when I close the app?

pooja-shah, 2017-05-09:
Have you find any solution? I am facing same issue. It also happens when I force stop iOS application.

Albertazzi88, 2017-05-25:
@pooja-shah unfortunately I haven't found any good and definitive solution.
The only way now is to make the app "crach" when I swipe it out using stopWithTask = "true" (don't know why it crashes anyway!), in this way everything is closed but an error message appear after the app is swiped out.
This is certainly not the best way to act.