evothings/cordova-ble #138: Ble scan only works the first time

elhampour, 2017-04-11:
I have an application. in my app i scan for ble devices. but it only works in just one way,
when my phone Bluetooth is off and i scan for ble , the plugin asks user to turn on Bluetooth , and then i finds the ble. but next time it does not work. 

i mean it only finds the ble when the plugin asks user to turn on  Bluetooth .

does any one know the problem ?

storm1er, 2017-04-12:
what function do you use to search devices ?

elhampour, 2017-04-12:
evothings.ble.startScan(onDeviceFound, onScanError, options)

i did test this a lot. but it only works the first time. android 5.1.1

storm1er, 2017-04-12:
When using it 'stock', just console.log result
It call onDeviceFound like every ... 100ms or something, if there is a peripheral in sight, you get the same ph. in loop, endlessly.

elhampour, 2017-04-12:
the problem that i have , is it only works when my bl is off ,and the app automatically turns it on. after the first time i done get the any peripheral endlessly,

storm1er, 2017-05-01:
Did you call 'stopScan' before 'startScan' again ?