evothings/cordova-ble #143: RSSI read not working in real app

sunnymtl, 2017-06-07:
I'm able to read the RSSI value continuously through Evothing app (on Samsung S4 Android 5.0) from a connected device. But after I built the custom app with exactly the same code, the RSSI value won't show. I have tried the online phonegap builder and the command line cordova builder, and another phone (G3 Android 6.0), got the same result: NO RSSI read. Could anybody help? should I add something when I build the real app?  

Here is the method I used:
evothings.ble.rssi(
deviceHandle,
function (rssi) { console.log("RSSI=" + rssi },
function (err) { console.log("ERROR=" + err });

Thanks a lot!

tozymandias, 2017-06-08:
I am having the same problem.  This method used to work at some point, and then it stopped working for some reason.  Please let me know if you find a solution.  Thanks.

sunnymtl, 2017-06-08:
@tozymandias Sure. I have submitted a ticket to ask the team, but no answer yet. No idea why it worked with evothing studio but not the app.

acutetech, 2017-07-01:
There is a bug in ble.js for the RSSI call - at about line 980:
	exec(deviceOrHandle, success, fail, 'BLE', 'rssi', [objectHandle(deviceOrHandle)]);
should be:
	exec(success, fail, 'BLE', 'rssi', [objectHandle(deviceOrHandle)]);

acutetech, 2017-07-01:
I see this was also reported and the bug found in #135

socca1157, 2017-08-03:
Is this resolved?

sunnymtl, 2017-08-05:
I have tried other ble plugins, finally use this: https://github.com/randdusing/cordova-plugin-bluetoothle, which doesn't have major problems.