evothings/cordova-ble #145: iOS 11 connection loss

qpwr, 2017-09-29:
I @mikaelkindborg, are you aware of any issues with iOS 11 with the plugin? I get more and more reports from iOS 11 users telling me that the app looses the connection after about 60 seconds and then reconnects with intermittent disconnects. I currently don't have any further details as my dev environment is not yet set up for iOS 11 (to get some debug output), thats why I'm just asking if you're aware of anything like this.

mikaelkindborg, 2017-09-29:
@qpwr Hi, sorry to say I don't maintain the plugin any longer, so I cannot say anything about how it works on iOS 11.

qpwr, 2017-09-30:
oh, okay, thanks for letting me know @mikaelkindborg. Does this mean that the project is not maintained anymore at all?

mikaelkindborg, 2017-09-30:
@qpwr I am not aware of the plugin being actively maintained at present. Last commit was made on Nov 24, 2016. There are some updates in forks: https://github.com/evothings/cordova-ble/network As an alternative, you could have a look at Don Coleman's plugin to see if that would work on iOS 11: https://github.com/don/cordova-plugin-ble-central Hope this helps you to move forward.